AndrewNolte/vscode-system-verilog #2: Syntax highlighting error in module declaration

terrywwgit, 2024-05-20:
**Describe the bug**
When the variable type is declared in the module declaration port list, the last port name is not colored correctly, the name is the same color as the port type.

**Environment (please complete the following information):**
 - OS: Windows 11 pro 23h2
 - vscode:
Version: 1.89.1 (user setup)
Commit: dc96b837cf6bb4af9cd736aa3af08cf8279f7685
Date: 2024-05-07T05:13:33.891Z
Electron: 28.2.8
ElectronBuildId: 27744544
Chromium: 120.0.6099.291
Node.js: 18.18.2
V8: 12.0.267.19-electron.0
OS: Windows_NT x64 10.0.22631

- vscode-system-verilog
v0.9.10

**Steps to reproduce**
Steps to reproduce the behavior:
1. create module declaration
2. terminate last port line with newline

---
  output **fsk_sym_o**, 
  output fsk_sym_n_o
);
---
if the ");" is moved to the end of the port line "output fsk_sym_n_o);" the problem does not occur.

**Log**
Attach the log generated while following the above steps

**Expected behavior**
port type and name should have different color.

**Actual behavior**
port name is same as port type

**Additional context**
Add any other context about the problem here.

AndrewNolte, 2024-05-21:
Thanks for the report. I've noticed a lot of these too, and they were common bug reports on upstream. I think I might copy in the config from https://github.com/eirikpre/VSCode-SystemVerilog, however In my opinion it makes the colors a bit weird, even though they are often (I'm assuming) more correct. It looks like they copied these on upstream. 

I'm going to try to find some time to play around with these but it's not a huge priority to me atm. If anyone's interested in experimenting, it's the tmLanguage files, more info at https://code.visualstudio.com/api/language-extensions/syntax-highlight-guide

AndrewNolte, 2024-08-19:
I updated the syntax highlighting in the recent update, do you want to check if this is still an issue?

terrywwgit, 2024-08-19:
Yes, it may be a little bit. I have a tapeout thursday, so real busy and
avoiding environment changes.

terrywwgit, 2024-10-10:
No longer an issue. I am now running the latest code.

closed 2024-10-10 by terrywwgit