AndrewNolte/vscode-system-verilog #6: [BUG] Include Globs are restricted to paths inside workspace

rahultadak, 2024-06-12:
It seems that the extension parses the include globs and expects the paths provided in the include globs to be inside the workspace always. This is not necessarily the case in projects where we refer to external IPs.

Environment
 - OS:
 ```
NAME="Rocky Linux"
VERSION="8.9 (Green Obsidian)"
ID="rocky"
ID_LIKE="rhel centos fedora"
VERSION_ID="8.9"
PLATFORM_ID="platform:el8"
PRETTY_NAME="Rocky Linux 8.9 (Green Obsidian)"
ANSI_COLOR="0;32"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:rocky:rocky:8:GA"
HOME_URL="https://rockylinux.org/"
BUG_REPORT_URL="https://bugs.rockylinux.org/"
SUPPORT_END="2029-05-31"
ROCKY_SUPPORT_PRODUCT="Rocky-Linux-8"
ROCKY_SUPPORT_PRODUCT_VERSION="8.9"
REDHAT_SUPPORT_PRODUCT="Rocky Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="8.9"
``` 
 - config: ``` {  "verilog.ctags.indexAllIncludes": true,
  "verilog.excludeGlob": "**/work/*",
  "verilog.lint.slang.path": "slang",
  "verilog.lint.slang.args": "--error-limit 200",
  "verilog.svFormat.verible.path": "verible-verilog-format",
  "verilog.ctags.path": "ctags",
  "verilog.includeGlobs": [
    "/path/to/uvm-1.2/*.{sv,svh}",
  ],
} ```

Steps to reproduce:
- Open Workspace
- Run re-index step- 

Logs:  (Get logs from Output > verilog)
```
2024-06-12 09:51:40.162 [info] [ctags] indexing 0 .svh files
2024-06-12 09:51:40.162 [info] [ctags] indexed 0 .svh files
2024-06-12 09:51:40.162 [info] ctags index includes finished
2024-06-12 09:51:40.190 [info] [index] indexing 18 files
2024-06-12 09:51:40.190 [info] index files finished
```

AndrewNolte, 2024-06-24:
Thanks for the report! This should be fixed in the most recent update, please let me know if it works for you! One note: because the filesystem glob is used instead of vscode's, the .{svh,sv} syntax doesn't work, or at least for the OS I tested on.

AndrewNolte, 2024-06-27:
No that won't really help, it just returns the same path, I believe either because it's a glob or because it's outside of the workspace. The documentation on that function isn't too clear, it seems to contradict itself

closed 2024-06-27 by AndrewNolte