AndrewNolte/vscode-system-verilog #9: [BUG] Verible format fails

gtaylormb, 2024-08-14:
**Describe the bug** - A clear and concise description of what the bug is.

Environment
 - OS:  
 ```
Operating System: Rocky Linux 9.4 (Blue Onyx)       
Kernel: Linux 5.14.0-427.28.1.el9_4.x86_64
```
 - config: 
 ``` 
    "verilog.ctags.path": "/usr/local/bin/ctags",
    "verilog.ctags.indexAllIncludes": true,
    "verilog.svFormat.formatter": "verible-verilog-format",
    "verilog.verilogFormat.formatter": "verible-verilog-format",
    "verilog.svFormat.verible.path": "/opt/verible/bin/verible-verilog-format",
    "verilog.verilogFormat.veribleFormatter.path": "/opt/verible/bin/verible-verilog-format"
 ```

Steps to reproduce:
 - Right clock in document, select Format Document

Logs:  (Get logs from Output > verilog)
```
2024-08-14 15:11:29.047 [info] [svFormat] formatting /home/gtaylor/git/fpga/src/clock_crossing/two_stage_synchronizer.sv
2024-08-14 15:11:29.047 [info] [svFormat] [verible] Temp file created at:/tmp/verible-verilog-format-c04d7d624381cefe245a29da3e69dfad.tmp.sv
2024-08-14 15:11:29.047 [info] [svFormat] [verible] Executing command: /opt/verible/bin/verible-verilog-format  --inplace /tmp/verible-verilog-format-c04d7d624381cefe245a29da3e69dfad.tmp.sv
2024-08-14 15:11:29.048 [error] [svFormat] [verible] Error: Command failed: /opt/verible/bin/verible-verilog-format  --inplace /tmp/verible-verilog-format-c04d7d624381cefe245a29da3e69dfad.tmp.sv
<empty-filename>: No such file or directory
```

AndrewNolte, 2024-08-19:
I've seen this before actually, and I'm not sure what causes it. I can experiment with having verible do the in-place formatting instead of creating the temporary file. That part wasn't written by me, but I assume they did it because not all formatters support inplace formatting.

AndrewNolte, 2024-08-19:
Contributions are welcome too! I don't think this should be too difficult for a starter task

AndrewNolte, 2024-08-26:
Should be fixed in 0.9.17, lmk if it works for you

closed 2024-08-26 by AndrewNolte

gtaylormb, 2024-10-01:
This unfortunately doesn't seem to work. I'm getting a different error now:

```
2024-10-01 11:24:11.879 [info] [svFormat] formatting /home/gtaylor/git/fpga/src/clock_crossing/two_stage_synchronizer.sv
2024-10-01 11:24:11.879 [info] [svFormat] [verible] Executing command: /opt/verible/bin/verible-verilog-format  /home/gtaylor/git/fpga/src/clock_crossing/two_stage_synchronizer.sv
2024-10-01 11:24:11.879 [error] [svFormat] [verible] Formatting failed: Error: Command failed: /opt/verible/bin/verible-verilog-format  /home/gtaylor/git/fpga/src/clock_crossing/two_stage_synchronizer.sv
--inplace required for multiple files.
```

I've tried setting the option to Run at file location, as well as setting `--inplace` as arguments to the tool, but they don't seem to have any effect either way (`--inplace` doesn't seem to get passed at all). I suspect if `--inplace` actually got passed it might work.

AndrewNolte, 2024-10-02:
Just pushed an update to the formatting implementation, can you try again? And --inplace only overwrites the file if there are absolutely no errors, so that's why I don't use that. It should pass it through if given though, which would result in an empty output, and so no format. Also, I'd say don't use run at location, it doesn't really make sense to have that for a formatter, especially if you reference say a flagfile in the args.

gtaylormb, 2024-10-07:
Now the file completely disappears after I run the command:

```
2024-10-07 09:19:15.601 [info] [svFormat] formatting /home/gtaylor/git/fpga/src/clock_crossing/two_stage_synchronizer.sv
2024-10-07 09:19:15.601 [info] [svFormat] [verible] Executing command: /opt/verible/bin/verible-verilog-format  -
```

AElimrani, 2025-05-08:
Sorry for bumping a closed issue but I wasn't sure if this is the same bug. It seems like I have a similar issue (nothing is passed to verible). I have the same settings as @gtaylormb:
`2025-05-08 17:12:45.432 [info] [verilogFormat] [veribleFormatter] Executing command: /Users/AnassElimrani/git_repos/fpga_projects/tools_bins/verible3986/verible/verible-verilog-format.exe  - `