AndrewNolte/vscode-system-verilog #10: Requesting support for pre-lint environmental configuration

RickyTino, 2024-08-16:
Hi!

I have installed slang under ~/.local/bin, but I failed to run slang lint when I am using this extension. The log says "Cannot find libatomic.so.1".  I googled, discovering that slang needs GCC-11 library, and I have GCC 11 installed but it is not initially included in $PATH. The problem is, I have no root access on the server, so I can only add GCC 11 to $PATH using .bashrc or "module load" (which is kind of our admin's setting), but none of them works when the extension executes slang. Don't know why, probably because it is using a non-interactive shell or something. 

I also tried mshr-h's [vscode-verilog-hdl-support](https://github.com/mshr-h/vscode-verilog-hdl-support), and discovered a workaround, by adding extra commands in front of the linting path, like: 
`"verilog.linting.path": "module load gcc/gcc-11.1.0; ~/.local/bin"`
But the same thing would not work on this extension, throwing some exception like "ENOENT". I read both of your code and discovered that you guys seem to be using different way to execute the linting command. ( I am not quite sure about that, since I am only an HDL guy)

So is it possible to add an option in your extension's settings to support some pre-lint environmental configuration? That would be really helpful for developers like me who has no root permission and cannot modify the $PATH. 

Thanks!

AndrewNolte, 2024-08-26:
So first, the slang path is meant to point at the binary, not the containing folder. And we want to avoid having the lint shell out every time it's run. So it'll try to get slang from the path on startup, and just use that. What will probably work for you is making a `slang.sh` file containing `module load gcc/gcc-11.1.0; ~/.local/bin/slang`

RickyTino, 2024-08-27:
Thank you so much for the reply! That works for me.

closed 2024-08-27 by RickyTino