AndrewNolte/vscode-system-verilog #12: moduleendmodule in comments

rskvprd, 2024-10-09:
I think that in [systemverilog.configuration.json](https://github.com/AndrewNolte/vscode-system-verilog/tree/main/languages/sv/systemverilog.configuration.json) 
 
```json
"autoClosingPairs": [
    { "open": "(", "close": ")" },
    { "open": "[", "close": "]" },
    { "open": "{", "close": "}" },
    { "open": "\"", "close": "\"", "notIn": ["string"] },
    { "open": "module", "close": "endmodule" }
  ],
```
should be something like that (I don't know what `"notIn"` accepts):
```json
"autoClosingPairs": [
    { "open": "(", "close": ")" },
    { "open": "[", "close": "]" },
    { "open": "{", "close": "}" },
    { "open": "\"", "close": "\"", "notIn": ["string"] },
    { "open": "module", "close": "endmodule", "notIn": ["comment", "string"]}
  ], 
```

I often used "module" in my comments like `// This module doing this and that and so on` and when I type `// This moduleendmodule` appears. I have to erase `endmodule` in comments every time :(

Thanks!

rskvprd, 2024-10-09:
Or maybe remove "moduleendmodule" from autoClosingPairs?

rskvprd, 2024-10-09:
So, I checked out that "notIn" accepts only "string" :(.
I think we need to remove it because of snippets already exists for this autocompletion

AndrewNolte, 2024-10-10:
It should be detecting that it's in a comment and not doing that. I can take a look soon. Yeah these and snippets to kind of overlap, I haven't really taken a look at those since forking.

AndrewNolte, 2024-10-14:
I got a fix for it, yeah just removed `{ "open": "module", "close": "endmodule" }`. Planning on releasing a patch update tomorrow with it, thanks for the contribution!

AndrewNolte, 2024-10-24:
Should be fixed in the latest patch!

closed 2024-10-24 by AndrewNolte