AndrewNolte/vscode-system-verilog #14: Exclude certain directories

terrenay, 2024-11-22:
Hi, first of all thanks for this useful tool! I'm not sure if this counts as a bug or as a feature request, I hope I chose correctly.

## Background

I'm working in an environment where I have a `src` folder which contains all the source code of my project. At build time, a new directory is created in the project's root, which contains (inside a subdirectory) a copy of all source files of my project as they were at the time the build started.

## Problem

The problem is that when I build and then continue working on the project, let's say to modify the types of some port of a module, the tool apparently gets confused which version of the module is being referenced: The one in the `src` folder (which is what I want) or the one in the build directory? For reference, I am using slang, verible-verilog-ls and ctags, and everything works fine unless the build directory is present. But once it is present, it appears to be quite random whether the module definition is taken from `src` or from `build`. This affects things like definitions upon hovering, slang error messages, etc.

## Example

The following excerpt from the extension's output illustrates what I mean. The module sequence_fifo which I instantiate is actually declared in `src/hdl/sequence_fifo.sv` and just happens to be copied to the `build_synth/...` directory when building:
```
src/hdl/bp_runlen_detector.sv:81:48: error: interface port 'o_data' not connected
  sequence_fifo #(.DEPTH(64), .AXI_WIDTH(512)) i_sequence_fifo(
                                               ^
build_synth/test_config_0/user_c0_0/hdl/ext/raw/sequence_fifo.sv:13:13: note: declared here
    AXI4S.m o_data,
            ^
```

## What I've tried

I have tried deleting the `.vscode-server/files` directory, reindexing, and renaming the build directory so that is comes before or after the src directory in the hierarchy (in the hope that the tools would consistently choose either the first or last version of a source file they encounter). I've also tried to exclude the directory from the file watcher in VSCode like so: 

```
"files.watcherExclude": {
    "**/build_synth/**": true
  }
```

I still get unpredictable behaviour. For some modules, the correct version is referenced, but not for others.

## Feature request

I wish there was a flag like `exclude-directories` one could set in `settings.json` which could get passed down to all tools used by the extension to ensure that certain directories are ignored. This would effectively avoid such a problem.

Thanks in advanced and please let me know if I have missed an existing way of dealing with this.

AndrewNolte, 2024-11-23:
Have you tried the config settings `verilog.includeGlobs` and `verilog.excludeGlobs`? The config options are documented at https://github.com/AndrewNolte/vscode-system-verilog/blob/main/CONFIG.md

terrenay, 2024-11-23:
Thanks! I totally missed the `verilog.excludeGlobs` option...
This solved the problem:

```
"verilog.excludeGlob": "*build*/**",
```

closed 2024-11-23 by terrenay