AndrewNolte/vscode-system-verilog #15: Fading unused variables

rskvprd, 2024-11-28:
Fade unused variables reported by slang

Currently unused variables reported as warnings by slang (as expected)
[image]

but maybe it can be faded like in other languages?
[image]

And i think if this feature is possible then it should be disabled when Editor: Show Unused is disabled
[image]

Thanks!

AndrewNolte, 2025-01-02:
Thanks for the suggestion, it shouldn't be too hard to add this. Looking back at the vscode api, it looks like there's also related diagnostics support, which may improve the look of things like name collisions, which are parsed separately at the moment.

AndrewNolte, 2025-02-28:
Thanks for the PR! Going to merge in separately though. There's some issues with this approach- 
- there are some unused diags like "unused return", where it wouldn't make sense to fade out
- If the user does want these to be an error- that still needs to be an option, which means adding a config for this. For example if there's a check in CI for these

AndrewNolte, 2025-02-28:
I think for the second point actually, what we could do is downgrade them to hint iff the severity is warning