AndrewNolte/vscode-system-verilog #17: [BUG] Linter Warnings/Errors Not Properly Marked in Multi-Level Directories

AdorableMiku1015, 2025-02-20:
Verilator sometimes outputs relative file paths rather than absolute file paths, causing the extension failed to mark linter warning/errors.
I'm trying to fix this bug.

Environment
OS: Windows 10
Linter: Verilator 5.020

Steps to reproduce:
 - step 1 Open a system verilog project folder with VSCode. The .sv source files shall distributed in multi-level directories.
 - step 2 Select a .sv file as top module.
 - step 3 Open the top module source file, and observing that no warnings/errors are successfully marked.
 - step 4 Check the extension outputs and find that verilator has outputed relative file paths, causing the extension failed to work.

Logs: 
```
2025-02-20 17:06:35.245 [error] [lint] [verilator] Error: Command failed: C:/verilator/verilator_bin.exe --lint-only +incdir+c:\Users\LJJ\Desktop\FPGA_SRC\FPGA-CAN\RTL -j 6 --sv --lint-only --language 1800-2017 --Wall -Wno-BADSTDPRAGMA -Wno-MULTITOP --timing +1364-2005ext+v -F verilator.do c:\Users\LJJ\Desktop\FPGA_SRC\FPGA-CAN\RTL\can_top.v
%Warning-WIDTHEXPAND: RTL/sub\can_level_bit.v:65:23: Operator ASSIGNDLY expects 17 bits on the Assign RHS, but Assign RHS's CONST '8'h0' generates 8 bits.
                                                   : ... note: In instance 'can_top.u_can_level_packet.u_can_level_bit'
   65 |         adjust_c_PBS1 <= 8'd0;
      |                       ^~
                      RTL/sub\can_level_packet.v:65:1: ... note: In file included from 'can_level_packet.v'
                      c:\Users\LJJ\Desktop\FPGA_SRC\FPGA-CAN\RTL\can_top.v:138:1: ... note: In file included from 'can_top.v'
                      ... For warning description see https://verilator.org/warn/WIDTHEXPAND?v=5.020
                      ... Use "/* verilator lint_off WIDTHEXPAND */" and lint_on around source to disable this message.
%Warning-WIDTHEXPAND: RTL/sub\can_level_packet.v:42:29: Operator ASSIGN expects 2 bits on the Assign RHS, but Assign RHS's CONST '1'h0' generates 1 bits.
                                                      : ... note: In instance 'can_top.u_can_level_packet'
   42 | initial {tx_done, tx_acked} = 1'b0;
      |                             ^
                      c:\Users\LJJ\Desktop\FPGA_SRC\FPGA-CAN\RTL\can_top.v:138:1: ... note: In file included from 'can_top.v'
%Warning-WIDTHEXPAND: RTL/sub\can_level_packet.v:144:29: Operator ASSIGNDLY expects 2 bits on the Assign RHS, but Assign RHS's CONST '1'h0' generates 1 bits.
                                                       : ... note: In instance 'can_top.u_can_level_packet'
  144 |         {tx_done, tx_acked} <= 1'b0;
      |                             ^~
                      c:\Users\LJJ\Desktop\FPGA_SRC\FPGA-CAN\RTL\can_top.v:138:1: ... note: In file included from 'can_top.v'
%Warning-WIDTHEXPAND: RTL/sub\can_level_packet.v:154:29: Operator ASSIGNDLY expects 2 bits on the Assign RHS, but Assign RHS's CONST '1'h0' generates 1 bits.
                                                       : ... note: In instance 'can_top.u_can_level_packet'
  154 |         {tx_done, tx_acked} <= 1'b0;
      |                             ^~
                      c:\Users\LJJ\Desktop\FPGA_SRC\FPGA-CAN\RTL\can_top.v:138:1: ... note: In file included from 'can_top.v'
%Error: Exiting due to 4 warning(s)

2025-02-20 17:06:35.246 [info] [lint] [verilator] cleared diags
```

AdorableMiku1015, 2025-02-21:
verilator.do content:
```
-y RTL
-y RTL/sub
```

AndrewNolte, 2025-02-28:
Thanks for the issue and PR! I just triggered a release, which should have the fix in it. I was able to move the fix into the base linter, so this should improve other linters as well! And you actually shouldn't need to put in -y; the extension does that for you by creating a temporary directory with symlinks to all of your sv files.

AdorableMiku1015, 2025-03-02:
> Thanks for the issue and PR! I just triggered a release, which should have the fix in it. I was able to move the fix into the base linter, so this should improve other linters as well! And you [...]

The use of symlinks works well, but in my case, the project is quite complex, featuring numerous source files that share the same name across different subfolders. Additionally, there are external IP files not located within the workspace folder that need to be passed to the linter. Therefore, I would prefer to use a separate "verilator.do" file to manage these references. Nonetheless, I truly appreciate this extension, and thanks for implementing the fix!

closed 2025-03-02 by AdorableMiku1015