AndrewNolte/vscode-system-verilog #22: how to enable the goto definition and Completions

yuqi-ali, 2025-04-09:
I chose the verible-verilog-ls as the Language Server and it. works, but not like the "Snappy Hover/Goto Definition on nearly every symbol, including in other files. Optional inlay hints for ports."described. 
This is my config:
`
{
    "verilog.ctags.path": "/opt/homebrew/Cellar/universal-ctags/p6.1.20250406.0/bin/ctags ",
    "verilog.ctags.indexAllIncludes": true,
    "verilog.index.indexAllIncludes": true,
    "verilog.languageServer.path": "/Users/yuqi/Downloads/verible-v0.0-3958-g7aae5c08-macOS/bin/verible-verilog-ls",
    "verilog.languageServer.server": "verible-verilog-ls",
    "verilog.lint.slang.enabled": false,
    "verilog.lint.verilator.enabled": true,
    "verilog.lint.verilator.path": "/opt/homebrew/bin/verilator"
}
`
Any Suggestions? Thanks

AndrewNolte, 2025-04-18:
Only one language server can be selected- the features described are for the ctags language server, so you should select that one. Beyond that, you may want to set includeGlobs and moduleGlobs. Check out https://github.com/AndrewNolte/vscode-system-verilog/blob/main/CONFIG.md for more info.

indexAllIncludes indexes the folders in `verilog.includes`, which is not set. You should put common include paths here for your project, which will also get passed to the lint tools.

I'm curious why you want to use verible-verilog-ls with this extension? You may want to see the discussion in https://github.com/AndrewNolte/vscode-system-verilog/issues/23 as well

yuqi-ali, 2025-04-23:
I notice that the Verilog/SystemVerilog Tools output err when start the vscode:
Pending response rejected since connection got disposed
The slang-server server crashed 5 times in the last 3 minutes. The server will not be restarted. See the output for more information.
```
error: no input files
[Error - 20:07:31] Server process exited with code 2.
[Error - 20:07:31] Server initialization failed.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Info  - 20:07:31] Connection to server got closed. Server will restart.
true
[Error - 20:07:31] slang-server client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
error: no input files
[Error - 20:07:31] Server process exited with code 2.
[Error - 20:07:31] Server initialization failed.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Info  - 20:07:31] Connection to server got closed. Server will restart.
true
[Error - 20:07:31] slang-server client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
error: no input files
[Error - 20:07:34] Server process exited with code 2.
[Error - 20:07:34] Server initialization failed.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Info  - 20:07:34] Connection to server got closed. Server will restart.
true
[Error - 20:07:34] slang-server client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 20:07:34] Restarting server failed
  Message: Pending response rejected since connection got disposed
  Code: -32097 
error: no input files
[Error - 20:07:34] Server process exited with code 2.
[Error - 20:07:34] Server initialization failed.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Info  - 20:07:34] Connection to server got closed. Server will restart.
true
[Error - 20:07:34] slang-server client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 20:07:34] Restarting server failed
  Message: Pending response rejected since connection got disposed
  Code: -32097 
error: no input files
[Error - 20:07:35] Server process exited with code 2.
[Error - 20:07:35] Server initialization failed.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 20:07:35] The slang-server server crashed 5 times in the last 3 minutes. The server will not be restarted. See the output for more information.
[Error - 20:07:35] slang-server client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 20:07:35] Restarting server failed
  Message: Pending response rejected since connection got disposed
  Code: -32097 
```
are there something else to do?

yuqi-ali, 2025-04-25:
unfortunately,it does not work again while restating the vscode
it seems that the ctags not work as OutLINE display No Symbols found in document '*.sv' when open sv file

AndrewNolte, 2025-04-25:
The `"verilog.languageServer.server": "slang-server"` setting should be set to ctags if you want the server features mentioned in the readme. The slang-server is still under development, and will be released in a separate extension. I'm not sure what you mean in the last couple of messages.

yuqi-ali, 2025-04-27:
Get It!This problem is fixed.Tanks

> The `"verilog.languageServer.server": "slang-server"` setting should be set to ctags if you want the server features mentioned in the readme. The slang-server is still under development, and will be released in a separate extension. I'm not sure what you mean in the last couple of messages.

yuqi-ali, 2025-04-27:
> all functions have worked. A small bugs:
> 
> ```
> a_pkt::struct_a_t a;
> ```
> 
> Snappy Hover/Goto Definition works for a
> 
> ```
> import a_pkt::struct_a_t;
> struct_a_t a;
> ```
> 
> Snappy Hover/Goto Definition not works for a

For This Question,Are there some config to do ?
And will this extension support struct field completion in the future?

AndrewNolte, 2025-05-06:
I've done a number of improvements to the completions here, but Ctags has an incomplete understanding of the structure, so it's a bit difficult to get all of this info for completions. I'm open to pull requests though if you'd like to look at `provideCompletionItems` in `src/analysis/CtagsServerComponent.ts`

I'm working on a new language server that goes much more in depth like this, and I'll make an announcement on this extension when that's released.