AndrewNolte/vscode-system-verilog #23: Request to add function to not index files

wpeet, 2025-04-12:
When using a Systemverilog language server, such as, verible-verilog-ls is it necessary to index all the files in the folder?

I tested an experiment where a disabled the indexComponent by commenting in out in the VerilogExtension class.
See the code snippet below. 
The GOTO Definition feature still worked.
In folders where there are a large number of files this seems to be a wasteful step.
I suggest a configuration option to skip the indexing

export class VerilogExtension extends ActivityBarComponent {
  /// Important components to load first
  **_// FIXME: index: IndexComponent = new IndexComponent()_**
  ctags: CtagsServerComponent = new CtagsServerComponent()

AndrewNolte, 2025-04-18:
Thanks for the feedback! So this should be influenced by `includeGlobs` and `indexAllIncludes` configs, however indexAllIncludes wasn't being checked in one of the callsites- I'll fix that.

Aside from this however- I'm curious what you're getting out of this extension over the upstream one. Much of the work done on this project is built on the ctags language server, which is implemented with the vscode api. The [upstream project](https://marketplace.visualstudio.com/items?itemName=mshr-h.VerilogHDL) had added other language servers, which to me seemed best provided by their own dedicated extension.

I'm considering refactoring the language server component to only allow for selecting one, which would implicitly disable this indexing that's used by the ctags language server. I'm curious if you run with multiple sv language servers, and if so what benefit you get out of it. If it's just lint messages from verible- perhaps it's best we add that as a linter, or a linter in the sense that it runs verible-ls configured to only show diagnostics (lint) messages. 

To easily see the difference, you can try running just the verible extension, and running this extension with ctags instead of verible

wpeet, 2025-04-19:
Making the index "component part of the ctags language server, so it only indexes if it's selected" makes sense to me.

I did not realize that this extension was built upon the [Verilog-HDL/SystemVerilog/Bluespec SystemVerilog extension](https://marketplace.visualstudio.com/items?itemName=mshr-h.VerilogHDL). I had tried the Verilog-HDL/SystemVerilog/Bluespec SystemVerilog extension, but it did not work well, because I did not have universal-ctags installed.

Refactoring of the language server component to only allow for selecting one makes sense to me. Running with multiple sv language servers does no make sense to me.

AndrewNolte, 2025-05-06:
I'm curious, in your logs, did you see something that said "ctags disabled, not indexing includes"? If so, the change I'm making would be moot.

And did you notice a difference when you commented it out? The file symlinking is still required for linting to work, so I'm wondering if that's the only step you were running, and if so why it was slow. It runs asynchronously as well, so it shouldn't affect other parts, unless you have some build directories with sv files that you didn't exclude.